MOLAorg/mola_lidar_odometry #4: ros2 launch of mola_lidar_odometry fails

sreevan, 2024-09-27:
mola-cli is failing to launch in ros2 and i am unable to find the reason for this. All i get is the error as in the screenshot and i have tried searching for any log of the error but could not find any. Can i please request help to identify the cause of this problem and i am sure it would be something to fix in my machine. Also, rarely it does launch successfully and molaviz pops up with the right scan and map display.

[image]

jlblancoc, 2024-09-27:
I guess your arch is arm64, right? (run ``uname --machine``)
It also looks like you have compiled everything from sources... have you tried installing via "apt", just for testing? Or is that not possible on your Jetson?

The only way I think of to find out more details, is to build with debug info: 

    colcon build --symlink-install  --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo

then enable core dumps, and after the crash, run ``coredump gdb`` and run ``bt`` to see where is actually happening the crash.

You can also run mp2p_icp unit tests just to verify that part is not the problem: 

    cd ros2_ws
    . install/setup.bash 
    cd build/mp2p_icp/ && make test

sreevan, 2024-09-28:
Thanks for the response. Yes it is aarch64 Jetson Tx2. I am unable apt install or use rosdep since i have built almost everything from sources (opencv, gtsam, boost, tbb etc) including ros2. Many libraries do not offer nightly builds for this cpu architecture. 

I had built mola exactly like as you advice 

```shell
colcon build --symlink-install  --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo
``` 
and was trying to get the backtrace but for the life of me could not get the coredump using gdb from ros2 launch with bt always saying no stack. 

When i run 
```shell
mola-cli --ros-args /media/nvidia/Expansion/sources/ros2_humble/install/mola_lidar_odometry/share/mola_lidar_odometry/mola-cli-launchs/lidar_odometry_ros2.yaml
```
it works everytime properly with the other ros nodes running in a different terminal feeding the lidar data. 
But, when i launch it through ros2 it fails most of the times and comes up properly once in a while.

And, all of the mp2p_icp tests passed
[image]

jlblancoc, 2024-09-28:
Thanks, that last call stack was really clarifying (once analyzed carefully...). 

I think it's fixed now after this commit: https://github.com/MOLAorg/mola/commit/f4340c6572fadf353f16cfd0522e88c3a0b43ecc
Please, update that repo (MOLAorg/mola), rebuild, and re-test it...

closed 2024-09-29 by sreevan