MOLAorg/mola_lidar_odometry #5: Knowledge needed

sreevan, 2024-09-29:
I am sorry to raise an issue on this but i am totally lost in the documentation and code when trying to figure out how to get a map and occupancy grid that is displayed on molaviz to be published onto ros2. With jlblancoc's help i now have good odometry on ros but as next step looking for map & OccGrid. I had been using one of the popular visual SLAM earlier but after a refresh of OS & ros2, it's various nodes are hogging CPU and also it is a bit unreliable based on day & night. I discovered this wonderful repo and decided to move to this and had been setting it up for a month now to the successful stage of lidar odometry but stuck and unable to move to the next steps without /map & /occgrid.

jlblancoc, 2024-09-29:
Hi @sreevan : 

During map building, if launched via `ros2 launch...`, there will be no gridmap, only point clouds as the actual metric map is an voxelmap (3D, no 2D), for which there's no direct equivalent AFAIK as ROS messages.

After map building ends and you have a .simplemap, you can then follow  [this tutorial](https://docs.mola-slam.org/latest/building-maps.html), specially sections 3 and 4. 
In section 4, you have an example (second tab) on how to generate a 2D gridmap. 

Build the metric map (the ".mm" file in the MOLA framework), inspect it with mm-viewer to see it's as expected, then publish it as a ROS message using `mrpt_map_server` (section 5 of the tutorial). 

Then, you can use the published map for localization, using any other package that expects a gridmap, or our mrpt_pf_localization; tutorial [here](https://docs.mola-slam.org/latest/localization.html#localization-with-particle-filtering).

Hope it helps!

sreevan, 2024-09-30:
Thanks very much @jlblancoc for the thoughtful response. Unfortunately, it actually was not what i was expecting. I think i will probably explain the outcome instead of intermediate steps so that you can guide me better. I am building a 4 wheel robot and in my ubuntu 18.04 & ros 1 days, i was able achieve quite a bit of steps towards my end goal. Had success on the dynamically adjusting mapping of the place, autonomous navigation to points, programmatically supplying nav goals over ros etc on the topic of physical movement of the robot. Now, all that progress has been set back after I started upgrading to ros2 for which i reflashed my jetson computers and started from the begining hoping all those libraries that i have integrated will work exactly like that (or even better) with latest code but to my dismay, there had been too many stumbling blocks one of which was odomtery. As mentioned earlier i am trying to move away from the visual slam package whose RGBD odom or icp odom could never be perfected like I was able to before. So, stumbled upon mrpt and mola, and with your help was able to get the odom perfection using mola lidar odomtery. I am now trying to work out as to how these packages can be used to get back to that state of the robot navigation I was long back. So, in a nut shell, looking for autonomous navigation of any environment (indoor only) with goal poses over ros. Would like to avoid two steps of build a map and then localize on a static map in trying to get that outcome. Would that be something that i can ask for help on?

jlblancoc, 2024-10-03:
oh, I see! So if you need a full navigation (localization + motion planning) stack, MOLA is only about the SLAM / localization part. 
We have [mrpt_navigation](https://github.com/mrpt-ros-pkg/mrpt_navigation) but probably Nav2 has tons more tutorials and features and might be the way to get into autonomous navigation.

In principle, MOLA output (/odom, /tf,...) should work with Nav2 (although, we have not setup a tutorial on that yet...)

sreevan, 2024-10-05:
Thanks very much for your advice, had been using ros navigation earlier and since i wanted a change in the SLAM library i thought i could use something that has better compatability and easy to set up. I will start learning nav2, gathered it is a bit different and needs a bit of reading & experimentation. Shall do that. Thanks again for all quick responses in this issue and the other one. Cheers.

closed 2024-10-05 by sreevan