MOLAorg/mola_lidar_odometry #6: "Y" axis values wavering

sreevan, 2024-11-02:
Hi,

I have been trying to get the navigation stack work with the lidar odometry from MOLA but there were errors from nav2 stack saying it was not able to find a right trajectory ("no valid trajectories of 0!). Few issues/call for help in ros forum and stack overflow mentioned that the usual cause is that a specific sensor value is noisy. After some research and watching the tf frames closely i found that the y axis values are fluctuating a bit badly from lidar odometry. Not sure of the reason and so seeking help. I have attached the log which my code generates by subscribing to the lidar odometry topic and logs the x, y & z parts of the pose. 
[python3_12029_1730548866341.log](https://github.com/user-attachments/files/17606845/python3_12029_1730548866341.log)

jlblancoc, 2024-11-04:
Hi @sreevan ! 

Some comments: 

1) On the noisy localization: The "dance" I can see in the MOLA Viz GUI is, indeed, undesirable, but shouldn't be IMO so large to become an issue with navigation. However, there are 2 main ways to improve it:

- By using a MOLA configuration with higher resolution, less data downsampling, etc. However, it seems you are using a 2D LiDAR, so downsampling should probably be already near the minimum... Using a particle filter ("amcl") typically also gives a similar amount of "pose noise". 

- By fusing & filtering information: One could use an ekf_node or alike. I think Steve has a great tutorial [here](https://docs.nav2.org/setup_guides/odom/setup_odom.html). We still don't have an equivalent ready-to-launch demo using MOLA, but it should exist soon in the future (!). 

2) What I think is your real problem, as far as I can tell, is the fact that the TF arrows "base_link -> odom" in RViz flicker (!). It seems to me that there are two entities publishing that pose transform. 

Check if wheels odometry are being published by some other node. Then, LiDAR odometry /tf frame should not be also named `odom` to avoid a collision. 

If you look at this line of MOLA LO params file: 
https://github.com/MOLAorg/mola_lidar_odometry/blob/develop/mola-cli-launchs/lidar_odometry_ros2.yaml#L50

you can see the tf frame name to be used. Try to change it (assuming you have mola_lidar_odometry built locally from sources), rebuild with colcon and check if that fixes the flickering. 

PS: TO-DO: 
- [ ] expose the odom frame as a parameter so it's more easily customizable from an external ROS2 launch file.

sreevan, 2024-11-05:
@jlblancoc Cannot thank you enough for pointing me in the right direction with your comment numbered 2. TF2 has always been my weakness that messes up things and my small brain could not visualise the tf tree properly. The "dance" is resolved now. Stuck with another problem of odom frame moving backwards but will have to ask help for that in ekf repo. Thanks again.

closed 2024-11-05 by sreevan