MOLAorg/mola_lidar_odometry #8: Livox MID360

David9696, 2025-01-06:
Hi, 

Thanks for providing the great package.

I have tried this package with my Livox MID360 3D lidar. But seems like it doesn't work. When the lidar is moving, the estimated odometry is not updated.

Is this package only support specific 3D lidar?

Hope you  could enlighten me .

Thanks.

jlblancoc, 2025-01-06:
Hi, 
I know of someones using it with Livox, but perhaps another model.
Please provide a short bag file for testing it on our side....

jlblancoc, 2025-01-10:
ping :face_in_clouds:  :-)

David9696, 2025-01-10:
Hi @jlblancoc ,

Thanks for your reply. I would prepare the bag file and send to you by this weekend. 

Thanks.

David9696, 2025-01-12:
Hi @jlblancoc ,

You may download the bag file from here: 

https://file.io/XsFMf8DFX5rL

In this bag file, the lidar is rotated for 360 degrees, on the spot.

Thanks.

jlblancoc, 2025-01-12:
Hi, 
It works. It's probably that you were missing the flag [to make it work without /tf](https://docs.mola-slam.org/latest/mola_lo_apps.html#mola-lo-gui-rosbag2): 

```bash
MOLA_LIDAR_TOPIC=/livox/lidar \
MOLA_USE_FIXED_LIDAR_POSE=true \
mola-lo-gui-rosbag2  /path/to/you/bag
``` 

[image]

David9696, 2025-01-14:
Hi @jlblancoc ,

Thanks for checking. Indeed, when the command run with bag file, it is working.

However, if i run it via the following ros2 command with live data:

ros2 launch mola_lidar_odometry ros2-lidar-odometry.launch.py    lidar_topic_name:=livox/lidar    ignore_lidar_pose_from_tf:=True

Seem like it doesn't work. The odometry is not updating. The icp quality is 0. I got the following errors continuously:

[mola-cli-1] [15:14:33.9371|WARN |LidarOdometry] Not able to use velocity motion model for this timestep (pathStep=19, timestamp=2025/01/14,08:14:33.818800 UTC)
[mola-cli-1] [15:14:36.0365|WARN |LidarOdometry] Not able to use velocity motion model for this timestep (pathStep=19, timestamp=2025/01/14,08:14:35.919886 UTC)
[mola-cli-1] [15:14:38.1320|WARN |LidarOdometry] Not able to use velocity motion model for this timestep (pathStep=19, timestamp=2025/01/14,08:14:38.019860 UTC)
[mola-cli-1] [15:14:40.1384|WARN |LidarOdometry] Not able to use velocity motion model for this timestep (pathStep=19, timestamp=2025/01/14,08:14:40.019972 UTC)

Hope you could enlighten me.

Thanks

jlblancoc, 2025-01-16:
I found the problem: this lidar driver seems to use timestamped points (the timestamp for each individual point within the PointCloud2 msg) in a way that is totally different to others: 
- Type:  FLOAT64
- Value: apparently, from [0 to 1e8] (??). Would be great to have a link to their docs explaining this. I don't understand why they use a float for this (sigh). 

Fixing this properly would imply adding a handler for this timestamp format within mrpt-ros2bridge. 

For the time being, you can just disable de-skewing and it works (fast motions and rotations may then lead to imperfections...): 

```bash
MOLA_SKIP_DESKEW=True \
  ros2 launch mola_lidar_odometry ros2-lidar-odometry.launch.py \
  lidar_topic_name:=/livox/lidar \
  ignore_lidar_pose_from_tf:=True \
  publish_localization_following_rep105:=False  # because you don't have a odom->base_link from wheeled robot
``` 

I'll leave this ticket open to remember about this timestamping issue.

David9696, 2025-01-22:
Hi @jlblancoc ,

Thanks for your reply.

I got the livox driver package from the following link:

https://github.com/Livox-SDK/livox_ros_driver2

I am not aware that their pointcloud topic format is different. It appears that the poincloud from livox lidar express the timestamps in nanoseconds. Could this be an issue?

Anyway, by using MOLA_SKIP_DESKEW=True, it works with livox mid360. 

Thanks

jlblancoc, 2025-01-26:
@David9696 Could you please share again the same test dataset, or a new one moving around with this LiDAR? The old link expired and I didn't keep a copy  🤦

David9696, 2025-02-02:
Hi @jlblancoc ,

You may download the bag file via the link below:

https://limewire.com/d/cdb39ec7-95fb-4345-9e83-f95b06cbd70f#F3H8_g7OeDO8DnSnaDrwFtJSJRGPRd9pilkHe6XVx5U

Thanks for looking into this issue.

David9696, 2025-02-14:
Hi @marmik18 ,

Noted with thanks. You may use the link below to download the bag file.

https://limewire.com/d/d2f13285-8d06-42af-ac00-a2302b611741#JipN5c_DfGJbZZ0ACzNA8v7_v1OXfW0vqll0lPRP_u8

marmik18, 2025-02-14:
Hey @David9696  I was able to run your bagfile with following command
```
MOLA_LOCALIZ_USE_REP105=False \
MOLA_USE_FIXED_LIDAR_POSE=True \
MOLA_ABS_MIN_SENSOR_RANGE=5 \
MOLA_SIGMA_MIN_MOTION=0.4 \
MOLA_LIDAR_TOPIC=/livox/lidar  \
mola-lo-gui-rosbag2 /path/to/rosbag
```

[image]

marmik18, 2025-02-14:
Hi @jlblancoc  
I am using the same sensor, apparently `Livox` uses a format `PointXYZRTLT`, which is different from others like `pcl :: PointXYZI` but `PointXYZRTLT` is still compatible with the ros2 message type `sensor_msgs::PointCloud2`.

Unfortunately, `pcl :: PointXYZI` is not supported by the `livox_ros_driver2` in `ros2`.
```
[livox_ros_driver2_node-1] error: message type 'pcl::PointCloud' is NOT supported in ROS2, please modify the 'xfer_format' field in the launch file
```

Here is the link to documentation:
https://github.com/Livox-SDK/livox_ros_driver2?tab=readme-ov-file#32-livox-ros-driver-2-internal-main-parameter-configuration-instructions

I would like to help on the issue, if you could point me in the right direction. 

Thank you and have a nice weekend :)

jlblancoc, 2025-02-15:
Awesome, thanks for sharing! 💯 

For the records: 
As a long term solution, it would be great to have Livox timestamps compatible with MOLA, so to convert them to "UNIX timestamp in seconds" or "second offsets" (around 0.0). 
The relevant implementation is [here](https://github.com/MRPT/mrpt/blob/develop/libs/ros2bridge/src/point_cloud2.cpp), although a transition for MRPT 3.0 is to delete that file and use instead [this repository instead](https://github.com/MRPT/mrpt_ros_bridge/blob/ros2/ros2bridge/src/point_cloud2.cpp) (which at present is an early WIP!).

marmik18, 2025-02-15:
Definitely, as a long term solution, it would be better to support Livox timestamps.
This is just a quick POC to verify whether the livox timestamps were the only problem. The livox sensor is a very different sensor compared to others, as the cloud is not repeatable.

Let me know if I can help in any way.

jlblancoc, 2025-04-16:
Awesome, Joaquín, thanks for the feedback! 💯

jlblancoc, 2025-04-23:
For the records: 
Livox timestamps are now 100% supported out of the box with MOLA v1.6.4 ([this commit](https://github.com/MOLAorg/mola/commit/fa0568107498a7bf02f3c8a33f580b4989b5b728)), so no need to tweaking its driver sources or disabling unskewing anymore. 

Explanation: 
- Handled at two places: live ROS2->MOLA bridge, and MOLA data reader from ROS2 bags. 
- How? If the input cloud has timestamps, and the time **span** within one scan is huge (>1e5 seconds), it is then assumed that those are actually nanoseconds, not seconds.

Hopefully, all LiDAR manufacturers in the future will remain using the same conventions, or we will run out of hacky patches ;-) 

PS: I still don't understand why Livox is using the costly "double" format to store nanosecond integers... (sigh)

closed 2025-04-23 by jlblancoc