MOLAorg/mola_lidar_odometry #10: Bad first ICP, re-starting from scratch with a new local map

StefanLandis, 2025-01-10:
Hi

It was not possible to generate a map. The map was repeatedly discarded.
Log output:
[15:35:59.3001|WARN |LidarOdometry] Bad first ICP, re-starting from scratch with a new local map 
[15:35:59.3001|WARN |LidarOdometry] Bad first ICP, re-starting from scratch with a new local map 

I found out that during initialization the values ​​from the bounding box were not yet available. This caused the variable radius to become a NaN in the doInitializeEstimatedMaxSensorRange method. After I caught the NaN in the doInitializeEstimatedMaxSensorRange method it worked fine.

[image]

jlblancoc, 2025-01-10:
Thanks for reporting! 

I can't imagine what might cause this, but, could you check if just [this code change](https://github.com/MOLAorg/mola_lidar_odometry/pull/11) fixes it?

StefanLandis, 2025-01-10:
I did it like this because I wasn't sure if the maxRange needs to get a new value.

[image]

I will also check your change.

StefanLandis, 2025-01-10:
Unfortunately the fix you suggested doesn't work for me.

[image]

closed 2025-01-11 by jlblancoc

StefanLandis, 2025-01-14:
Please excuse my late reply. And thank you for the fix.

Unfortunately I can't explain it either. Yes, it only happens during initialization.

I use the [Huraison Panter simulation](https://github.com/husarion/panther_ros) as sensor input. The point cloud generated by mola looks good. After I no longer had any NaN.

[image]

jlblancoc, 2025-01-14:
:cool: