MOLAorg/mola_lidar_odometry #9: Need help on integrating imu for lidar odometry

tenvinc, 2025-01-10:
Thank you for the wonderful framework. I have tried to run the code with the Mulran dataset, without the use for imu and the localization works perfectly. However, once I included the imu readings from the Mulran KAIST01 sequence, the pose calculated starts jumping around and localization fails.

I am currently using the tag 0.5.0 of the mola lidar odometry and the rest of the packages are 
[image]

I have tried debugging this on my own, by looking at the debug logs. I keep getting the message **"Ignoring IMU reading at t=1561000446.684496 due to missing former navstate estimation"**. I enabled the initial re-localization feature, the message went away momentarily, but after some time, the messages would appear again.

This is the case even after I disabled the gps readings so the module is purely reliant on the initial pose at (0, 0, 0), the lidar and the imu readings.

Have you faced similar problems?
[lidar_odom.log](https://github.com/user-attachments/files/18369771/lidar_odom.log)

jlblancoc, 2025-01-10:
Thanks, @tenvinc ! I can confirm I could replicate the issue, even for the current develop branch (!)... I'll investigate it and probably add a new CI test including IMU.

jlblancoc, 2025-01-10:
@tenvinc the fix (in the mola-state-estimation package) has been merged.

closed 2025-01-10 by jlblancoc

tenvinc, 2025-01-13:
> BTW: How do you have mola-1.5.1 in Humble via apt? Did you enable the "testing" repo instead of "stable", right?

yes, I enabled the "testing" repo, because I was having some issues with my custom dataset and wanted to see if adding IMU would improve the lidar odometry.