MOLAorg/mola_lidar_odometry #13: Localization in gazebo simulation

alkris-okr, 2025-01-23:
Hello, I’m new to working with ROS and I’m trying to create a 3D map of a Gazebo world for localization. I’m using RViz for visualization and providing pose estimation for localization. The problem I’m encountering is when I try to relocalize the robot—if I move it to a random position—rather than moving to the new location, a new map appears in the MM-Viewer, overlapping the previously loaded map. As I move the robot around, this new map keeps updating, even though ‘**_start_mapping_enabled_**’ is set to false. Since I’m still learning, I’m not entirely sure what’s going wrong. Just a few days ago, everything was working fine when I first built the MOLA package from source, and there were no issues with localization. However, after making changes to the lidar_odometry_ros2.yaml file, the issue started, and even after undoing those changes, the problem persists. I’m not sure if it’s related to the changes I made or something else. Any guidance would be greatly appreciated!

_**Here is the loaded map with localization disabled ---**_ 

[image]

_**Localization is enabled and robot has been moved to several random location ---**_

[image]

r-aguilera, 2025-01-23:
Hi @alkris-okr, I also had my map being updated even after I set `start_mapping_enabled` as false. Maybe #12 is related to your problem. If it solves/causes it, might be worth checking out!

alkris-okr, 2025-01-23:
Hello @r-aguilera, the new update resolved the issue. Thank you so much for you assistance and this wonderful framework.

closed 2025-01-23 by alkris-okr