MOLAorg/mola_lidar_odometry #14: Map generation and localization analysis

alkris-okr, 2025-01-30:
Hello,

I've been using this amazing repository for several weeks now, and I’ve been working on mapping a Gazebo environment and using that map for robot localization. Additionally, I’ve been analyzing the localization performance. Along the way, I’ve encountered several questions, and I would really appreciate it if anyone could help clarify them. I’m still a beginner, so I apologize if these questions seem trivial, and I’m grateful for any assistance you can offer.

I followed the steps in this [tutorial](https://docs.mola-slam.org/latest/tutorial-mola-lo-map-and-localize.html#tutorial-mola-lo-map-and-localize) for both mapping and localization.

Q1) My 3D map doesn’t appear as smooth as I expected. The map shown below seems to be incomplete or not fully aligned with the actual environment. During mapping, I also noticed that my robot wasn’t following the exact same path as in Gazebo; it was displaying some zigzag movements around the path. I thought this might be related to SLAM, but I couldn’t figure out how to resolve this issue.
<table>
  <tr>
    <th>Gazebo Environment</th>
    <th>3D MAP</th>
  </tr>
  <tr>
    <td>[image]</td>
    <td>[image]</td>
  </tr>
</table>

Q2) Below, you can see the trajectory of the estimated path compared to the ground truth. Is it normal to have this zigzag deviation from the ground truth, or is this some issue?
<table>
  <tr>
    <td>[image]</td>
  </tr>
</table>

Q3) During localization, I’m seeing Z coordinates, even though the robot is supposed to move in 2D. As shown in the graph below, there are fluctuations along the Z-axis, even though the ground truth is consistently at 0.
<table>
  <tr>
    <td>[image]</td>
  </tr>
</table>

Additionally, I’m seeing **_roll_** and **_pitch_** data being generated, while I was expecting to only receive _**yaw**_ data since the robot is meant to move in a 2D plane. Could this be the reason to the fluctuations in the APE data?
<table>
  <tr>
    <td>[image]</td>
    <td>[image]</td>
  </tr>
</table>

jlblancoc, 2025-02-03:
Hi! 

The `lidar3d-default` configuration was defined to provide a good performance on driving (outdoor) datasets, and an "ok" performance indoors. 
But users can and should tune it for each case to have better results.

I recommend trying two things in your case: 
- Give it a try to the 3D-NDT method. If you are using mola-lo-gui-* tools, see [this example](https://docs.mola-slam.org/latest/mola_lo_apps.html#mola-lo-gui-mulran) for the environment variables you can set. Play a bit with different `MOLA_LOCAL_VOXELMAP_RESOLUTION`.
- Launch the same `lidar3d-default` method, with smaller `MOLA_LOCAL_VOXELMAP_RESOLUTION` values, e.g. 

```
MOLA_LOCAL_VOXELMAP_RESOLUTION=0.50 \
ros2 launch ...
# or
MOLA_LOCAL_VOXELMAP_RESOLUTION=0.50 \
mola-lo-gui-rosbag2 ...
``` 
In theory, smaller numbers should give better results, at the cost of slower runs.

On the pure 2D motion: in the past there was an option to enforce a pure (x,y,phi) solution and enforcing zeros in Z, pitch, roll. But for real robots it doesn't make actually sense since we would always have small vibrations, etc. that make it better to allow ICP to estimate those small movements. In a simulator... yes, it might make sense to have such an option again :-)

jlblancoc, 2025-03-03:
For the records, there is now such a force 2D motion model again. 
If using the [ros2 node](https://docs.mola-slam.org/latest/ros2api.html) it can be enabled with `ros2 launch ... enforce_planar_motion:=True`.

closed 2025-03-03 by jlblancoc