MOLAorg/mola_lidar_odometry #15: Not able to use velocity motion model

JeremyLebon, 2025-02-04:
``I'm currently trying the mola lidar odometry package. I'm using the ROS2 Jazzy and Gazebo Harmonic to simulate a outdoor factory env.

The needed topics are received in the mola (docker container). 
When   `````launching` `ros2 launch mola_lidar_odometry ros2-lidar-odometry.launch.py ``

The viewer is started and some points are visible.

[image]

But after some (5 à 10 secs) I get the following output 

**[mola-cli-1] [14:15:59.1496|WARN |LidarOdometry] NaN bounding box for sensor point cloud. Using default sensor range.**
[mola-cli-1] [14:15:59.1682|INFO |mola::BridgeROS2:ros2_bridge] Subscribing to MOLA localization source module 'mola::LidarOdometry:lidar_odom'
[mola-cli-1] [14:15:59.1682|INFO |mola::BridgeROS2:ros2_bridge] Subscribing to MOLA map source module 'mola::LidarOdometry:lidar_odom'
[mola-cli-1] [14:15:59.1682|INFO |mola::BridgeROS2:ros2_bridge] Subscribing to MOLA relocalization module 'mola::LidarOdometry:lidar_odom'
[mola-cli-1] [14:15:59.1831|INFO |mola::BridgeROS2:ros2_bridge] Subscribing to MOLA map server module 'mola::LidarOdometry:lidar_odom'
[mola-cli-1] [14:16:07.0308|WARN |LidarOdometry] Not able to use velocity motion model for this timestep (pathStep=67, timestamp=1970/01/01,00:12:26.100000 UTC)
[mola-cli-1] [14:16:09.0748|WARN |LidarOdometry] Not able to use velocity motion model for this timestep (pathStep=67, timestamp=1970/01/01,00:12:28.100000 UTC)
[mola-cli-1] [14:16:11.1053|WARN |LidarOdometry] Not able to use velocity motion model for this timestep (pathStep=67, timestamp=1970/01/01,00:12:30.000000 UTC)
[mola-cli-1] [14:16:13.1152|WARN |LidarOdometry] Not able to use velocity motion model for this timestep (pathStep=67, timestamp=1970/01/01,00:12:32.000000 UTC)
[mola-cli-1] [14:16:15.1351|WARN |LidarOdometry] Not able to use velocity motion model for this timestep (pathStep=67, timestamp=1970/01/01,00:12:33.900000 UTC)
[mola-cli-1] [14:16:17.1857|WARN |LidarOdometry] Not able to use velocity motion model for this timestep (pathStep=67, timestamp=1970/01/01,00:12:35.900000 UTC)
 

What i see is that there is a warning  about `NaN bounding box`. When receiving the warning the warning message Not able to use velocity motion model is afterwards received. Any suggestion on what causes this?

jlblancoc, 2025-02-04:
If you have installed mola_lidar_odometry via ``apt install ...`` this is likely the same bug already solved in e9dfabdebdd894008bcb2a4003588d30201dd60d . 
It may take weeks for new versions to be available from the ROS build farm. 

You can verify if this is the case by either: 
a) remove mola_lidar_odometry with ``apt uninstall`` , then clone it and build with colcon; or
b) enabling [the testing repo](https://docs.ros.org/en/rolling/Installation/Testing.html).

JeremyLebon, 2025-02-06:
Thanks for the hint!

I rebuild from source (develop branch) and I still get the same warnings.

I bypassed/altered the radius value manual to 500m. 

````cpp
// check for NaN: See: https://github.com/MOLAorg/mola_lidar_odometry/issues/10
  if (radius != radius) {
    MRPT_LOG_WARN_STREAM("NaN bounding box for sensor point cloud. Using default sensor range.");
    radius = 500.0;
  } else {
    mrpt::keep_max(radius, params_.absolute_minimum_sensor_range);
  }
````
With the adjustment above the second warning `Not able to use velocity motion model for this`   takes longer to be trigger.   The problem is that the variable `hasMotionModel`  (line859) at some point state that there is no MotionModel any more.
````cpp
const bool hasMotionModel = state_.last_motion_model_output.has_value();
````
What could cause that there isn' t any MotionModel anymore?  I can see the following code is probably responsible for it.
````cpp
state_.last_motion_model_output =
    state_.navstate_fuse->estimated_navstate(this_obs_tim, params_.publish_reference_frame);
````
Any idea what could cause this reaction?

jlblancoc, 2025-02-07:
Could you try recording a bag where this happens so we can debug it offline?

jlblancoc, 2025-02-07:
PS: 

> The needed topics are received in the mola (docker container).

How did you enable the custom mola-gui from within the docker container? :-) I'm always troubling to achieve GUI stuff from within docker...

JeremyLebon, 2025-02-10:
> Could you try recording a bag where this happens so we can debug it offline?

See [link](https://we.tl/t-uYifDaYpCY) with 2 rosbags. Thanks for the feedback!

JeremyLebon, 2025-02-10:
> PS:
> 
> > The needed topics are received in the mola (docker container).
> 
> How did you enable the custom mola-gui from within the docker container? :-) I'm always troubling to achieve GUI stuff from within docker...

I work with a docker compose file. See configuration below.

```
services:
  mola-core:
    #restart: always
    build:
      context: ./
      dockerfile: ./docker/Dockerfile
      args:
        - RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
        - CYCLONEDDS_URI=file:///root/ros2_ws/cylconedds_config.xml
    network_mode: host
    stdin_open: true
    tty: true  
    runtime: nvidia
    environment:
      - DISPLAY=${DISPLAY}
      - QT_X11_NO_MITSHM=1
      - XAUTHORITY=${XAUTH}
      - NVIDIA_VISIBLE_DEVICES=all
      - NVIDIA_DRIVER_CAPABILITIES=all
      - ROS_DISTRO=jazzy
      - ROS_DOMAIN_ID=5
      - RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
      - CYCLONEDDS_URI=file:///root/ros2_ws/cylconedds_config.xml
    container_name: mola-core
    privileged: false
    cap_add:
      - NET_ADMIN
    volumes:
      - /tmp/.X11-unix:/tmp/.X11-unix:rw
      - ./mola_lidar_odometry:/root/ros2_ws/src/mola_lidar_odometry
    command: bash 
    deploy:
      resources:
        reservations:
          devices:
            - driver: 'nvidia'
              count: all
              capabilities: [gpu]

```

jlblancoc, 2025-02-11:
@JeremyLebon this was a curious case... perhaps because it seems a simulation, the automatic lidar range estimator does not catch up fast enough (the sensed range quickly increase from 50m to 200+ meters in the few frames), and the adaptive behavior of the default settings plays against accuracy in this case. 

You can run modifying the defaults like this and it works: 

```bash
MOLA_ABS_MIN_SENSOR_RANGE=200 \
MOLA_SIGMA_MIN_MOTION=0.5 \
MOLA_LIDAR_TOPIC=/lidar/points \
mola-lo-gui-rosbag2 rosbag2_2025_02_10-10_52_58/rosbag2_2025_02_10-10_52_58_0.mcap
```

[image]

In general, one can follow [the instructions](https://docs.mola-slam.org/latest/building-maps.html#see-the-inner-workings-of-icp) on how to visualize what's going on inside the ICP method to gain insight in what's the actual problem. Here, voxelization resolution, filtering near points, and the lack of distant points from the first frames all "confabulated" to take the ICP pairing ratio towards zero :-)

JeremyLebon, 2025-02-13:
Thanks for the quick response. The adjustments of the parameters seem to do the trick.

closed 2025-02-13 by JeremyLebon