MOLAorg/mola_lidar_odometry #16: Fails to build

dnovischi, 2025-02-13:
Building according to the [official docs](https://docs.mola-slam.org/latest/index.html#installing):

```~/ros2_mola_ws/src/mola_lidar_odometry/module/src/LidarOdometry.cpp:2176:6: error: ‘struct mola::LocalizationSourceBase::LocalizationUpdate’ has no member named ‘child_frame’
 2176 |   lu.child_frame = params_.publish_vehicle_frame;
      |      ^~~~~~~~~~~
gmake[2]: *** [CMakeFiles/mola_lidar_odometry.dir/build.make:76: CMakeFiles/mola_lidar_odometry.dir/module/src/LidarOdometry.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:126: CMakeFiles/mola_lidar_odometry.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
```

I stumbled across this while desperately trying to set base_footprint  (as in `map -> base_footprint -> base_link -> laser_link`), which for me simply did not work (neither by modifying environment variables nor by modifying the yamls w.r.t base_link). 

However, I can confirm that something like:
`ros2 launch mola_lidar_odometry ros2-lidar-odometry.launch.py lidar_topic_name:=laser_cloud  publish_localization_following_rep105:=False mola_lo_pipeline:=lidar2d.yaml` with tf2 set only with the base_link -> laser works when the mola packages are installed form apt manager.

Away, I'm stuck. Whats the proper way to indicate base_footprint for mola?

jlblancoc, 2025-02-13:
Hi, 

That build error is because that version of mola_lidar_odometry needs a more recent version of mola_bridge_ros2. Check if you have the latest, via apt, or building the MOLAorg/mola repo from sources.

To have base_footprint: You need to uncomment [these two lines](https://github.com/MOLAorg/mola_lidar_odometry/blob/develop/mola-cli-launchs/lidar_odometry_ros2.yaml#L66-L67) and put your values there.  Note that the tf tree will not be exactly as you depicted, since base_footprint will be a (leaf) child of base_link. But it shouldn't matter: any query to the tf2 system will report the correct transformations.

If you see it useful, feel free of opening a PR to expose those two lines via env variables, or even an additional ros2 launch argument (?).

dnovischi, 2025-02-13:
My bad, I had both from package and from source.

closed 2025-02-13 by dnovischi