MOLAorg/mola_lidar_odometry #17: Exception when activating base_footprint

dnovischi, 2025-02-13:
Hi,

I'm able to build, then run with: `ros2 launch mola_lidar_odometry ros2-lidar-odometry.launch.py lidar_topic_name:=laser_cloud ignore_lidar_pose_from_tf:=False    publish_localization_following_rep105:=False mola_lo_pipeline:="../pipelines/lidar2d.yaml"`

However, when I try to make it publish to tf the base_footprint, as in:

```
      # If not empty, the node will broadcast a static /tf from base_link to
      # base_footprint with the TF base_footprint_to_base_link_tf at start up.
      #base_footprint_frame: 'base_footprint'
      #base_footprint_to_base_link_tf: '[x y z yaw_deg pitch_deg roll_deg]'
      base_footprint_frame: 'base_footprint'
      base_footprint_to_base_link_tf: '[0 0 0.04 0 0 0]'
```

I get an exception that crashes the ros_bridge. Here's the verbose output with DEBUG level activated:

```
$ ros2 launch mola_lidar_odometry ros2-lidar-odometry.launch.py lidar_topic_name:=laser_cloud ignore_lidar_pose_from_tf:=False    publish_localization_following_rep105:=False mola_lo_pipeline:="../pipelines/lidar2d.yaml"
[INFO] [launch]: All log files can be found below /home/dan/.ros/log/2025-02-13-21-08-50-246076-dan-legion5-72802
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [mola-cli-1]: process started with pid [72803]
[INFO] [rviz2-2]: process started with pid [72805]
[mola-cli-1] [21:08:50.3530|INFO |MolaLauncherApp] Setting up system from YAML config... (set DEBUG verbosity level to see full config)
[mola-cli-1] [21:08:50.3531|INFO |MolaLauncherApp] Instantiating module `viz` of type `mola::MolaViz`
[mola-cli-1] [21:08:50.3545|INFO |MolaLauncherApp] Instantiating module `lidar_odom` of type `mola::LidarOdometry`
[mola-cli-1] [21:08:50.3547|INFO |MolaLauncherApp] Instantiating module `state_estimation` of type `mola::state_estimation_simple::StateEstimationSimple`
[mola-cli-1] [21:08:50.3548|INFO |MolaLauncherApp] Instantiating module `ros2_bridge` of type `mola::BridgeROS2`
[mola-cli-1] [21:08:50.3551|INFO |MolaLauncherApp] ╔═══════════════════════════════════════╦═════════════════════════════════╗
[mola-cli-1] [21:08:50.3551|INFO |MolaLauncherApp] ║  Entering main MOLA application loop  ║ > CTRL+C for mola-cli to quit < ║
[mola-cli-1] [21:08:50.3551|INFO |MolaLauncherApp] ╚═══════════════════════════════════════╩═════════════════════════════════╝
[mola-cli-1] [21:08:50.3552|DEBUG|mola::MolaViz:viz] Loading these params:
[mola-cli-1] ~  # none
[mola-cli-1] [21:08:50.3552|DEBUG|mola::LidarOdometry:lidar_odom] initialize_common(): subscribing to 'ros2_bridge' for raw sensor input.
[mola-cli-1] [21:08:50.3552|DEBUG|StateEstimationSimple] initialize() called with:
[mola-cli-1] name: state_estimation
[mola-cli-1] # This file holds parameters for mola::state_estimation_simple::StateEstimationSimple
[mola-cli-1] # 
[mola-cli-1] # # # # Refer to:
[mola-cli-1] # https://docs.mola-slam.org/latest/mola_state_estimators.html
[mola-cli-1] params:
[mola-cli-1]   # Optional initial guess for the twist (vx vy vz: m/s, wx wy wz: rad/s):
[mola-cli-1]   initial_twist:
[mola-cli-1]     - 0.0
[mola-cli-1]     - 0.0
[mola-cli-1]     - 0.0
[mola-cli-1]     - 0.0
[mola-cli-1]     - 0.0
[mola-cli-1]     - 0.0
[mola-cli-1]   # Maximum time since last observation to consider the validity of the velocity model:
[mola-cli-1]   # (KITTI360 test_3 has some blackouts >=1.0 seconds while angular acceleration is high, so
[mola-cli-1]   #  this threshold must be < 1.0 s for that dataset)
[mola-cli-1]   max_time_to_use_velocity_model: 0.75  # [s]
[mola-cli-1]   sigma_random_walk_acceleration_angular: 10.0  # [rad/s²]
[mola-cli-1]   sigma_random_walk_acceleration_linear: 1.0  # [m/s²]
[mola-cli-1] type: mola::state_estimation_simple::StateEstimationSimple
[mola-cli-1] verbosity_level: DEBUG
[mola-cli-1] 
[mola-cli-1] [21:08:50.3552|INFO |StateEstimationSimple] reset() called
[mola-cli-1] [21:08:50.3552|DEBUG|mola::LidarOdometry:lidar_odom] No SLAM back-end found in the system.
[mola-cli-1] [21:08:50.3552|INFO |mola::LidarOdometry:lidar_odom] Attached to a VizInterface module
[mola-cli-1] [21:08:50.3552|DEBUG|mola::MolaViz:viz] gui_thread() started.
[mola-cli-1] [21:08:50.3553|DEBUG|LidarOdometry] Loading these params:
[mola-cli-1] absolute_minimum_sensor_range: 20.0
[mola-cli-1] # Adaptive threshold, as in the KISS-ICP paper:
[mola-cli-1] adaptive_threshold:
[mola-cli-1]   alpha: 0.99
[mola-cli-1]   enabled: true
[mola-cli-1]   initial_sigma: 0.20
[mola-cli-1]   kp: 2.0
[... 267 lines cut ...]
[mola-cli-1] [3 ]     0x76584acdc253 
[mola-cli-1] [4 ]     0x76584a894ac3 
[mola-cli-1] [5 ]     0x76584a926850 
[rviz2-2] [INFO] [1739473730.476803409] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-2] [INFO] [1739473730.476946289] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-2] [INFO] [1739473730.505794422] [rviz2]: Stereo is NOT SUPPORTED
[mola-cli-1] [21:08:50.3686|DEBUG|LidarOdometry] ExecutableBase dtor called for module: `mola::LidarOdometry:lidar_odom`
[mola-cli-1] [21:08:50.3687|INFO |mola::LidarOdometry:lidar_odom timeLogger] dumpAllStats:
[mola-cli-1] ----------- mola::LidarOdometry:lidar_odom: MRPT CTimeLogger report ------------
[mola-cli-1]            FUNCTION                         #CALLS  MIN.T  MEAN.T MAX.T TOTAL 
[mola-cli-1] --------------------------------------------------------------------------------
[mola-cli-1] filterPointCloud_initialize                   1   3.2ms   3.2ms   3.2ms   3.2ms
[mola-cli-1] ----------- mola::LidarOdometry:lidar_odom: MRPT CTimeLogger report ------------
[mola-cli-1] [21:08:50.5552|INFO |MolaLauncherApp] Main SLAM/localization loop ended.
[mola-cli-1] [21:08:50.5552|INFO |MolaLauncherApp] shutdown(): Shutting down 4 module threads...
[mola-cli-1] [21:08:51.3557|DEBUG|StateEstimationSimple] ExecutableBase dtor called for module: `mola::state_estimation_simple::StateEstimationSimple:state_estimation`
[mola-cli-1] [21:08:51.3868|DEBUG|mola::MolaViz:viz] gui_thread() quitted.
[mola-cli-1] [21:08:51.3872|DEBUG|mola::MolaViz:viz] ExecutableBase dtor called for module: `mola::MolaViz:viz`
[mola-cli-1] [21:08:51.3873|INFO |MolaLauncherApp] shutdown(): Done.
[INFO] [mola-cli-1]: process has finished cleanly [pid 72803]
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[rviz2-2] [INFO] [1739473739.959350654] [rclcpp]: signal_handler(signum=2)
[INFO] [rviz2-2]: process has finished cleanly [pid 72805]

```

jlblancoc, 2025-02-14:
It's a bug in parsing the footprint -> base_link transformation. Look at the first ``==== MRPT exception ====``.

It's now fixed, you need to "git pull" both "mola" and "mola_lidar_odometry" (to see the new, correct format for ``base_footprint_to_base_link_tf`` (comma separated).

Note however that I think the actual pose is not being published for ``base_footprint => base_link`` according to RViz, please check if it's the same for you.

dnovischi, 2025-02-14:
It did the trick. Thank you!

Consider putting in parameters for the frames in the launcher yaml (or possibly in the pipelines, but  should normally trickle down to them). I say this because static tf's usally come from robot structural parameters and as such are solely part of robot builds (i.e. just inputs for both slam front-and back- ends).

jlblancoc, 2025-02-15:
With the last commit 9b2a8fcd2c166236e8f936655ee910682ee6411c we now have a ros argument to configure the base_footprint /tf. Also, in MOLAorg/mola, the order in which the /tf are published has been swapped to avoid warnings in RViz (note that the order of the /tf is swapped, but the transformation is also internally inverted, so the meaning of the parameter remains the same: transformation **from** base_footprint **to** base_link). 

Example: 

```bash
ros2 launch mola_lidar_odometry ros2-lidar-odometry.launch.py  \
    lidar_topic_name:=/xxxxx_points \
    mola_footprint_to_base_link_tf:='[0, 0, 0.20, 0, 0, 0]'
```

closed 2025-02-15 by jlblancoc