MOLAorg/mola_lidar_odometry #18: Robot begins to float (z-axis)

JeremyLebon, 2025-02-14:
When driving the robot in a straight line in gazebo (Harmonic) env. The robot begins to float (rise in height). When using another slam algorithm the robot stays on the ground. Any suggestions what could be cause? Or what config parameters could solve the problem?

[image]

See rosbag in [link](https://we.tl/t-w0PkIvlMyq) 

Thanks for the feedback

marmik18, 2025-02-14:
+1 even I have also faced this

jlblancoc, 2025-02-15:
I tried with NDT to see if it helps and it seems to mitigate it a bit, but still the +Z motion is not good enough. 

For the records: 
```bash
# Run with 3D-NDT method:
PIPELINE_YAML=$(ros2 pkg prefix mola_lidar_odometry)/share/mola_lidar_odometry/pipelines/lidar3d-ndt.yaml \
MOLA_LOCAL_VOXELMAP_RESOLUTION=5.0 \
MOLA_ABS_MIN_SENSOR_RANGE=200 \
MOLA_SIGMA_MIN_MOTION=0.5 \
MOLA_MINIMUM_RANGE_FILTER=3.0 \
MOLA_LIDAR_TOPIC=/lidar/points \
mola-lo-gui-rosbag2 rosbag2_2025_02_13-10_19_34_0.mcap
```

The origin of this problem seems to be that there are too few points on the floor (and they are not simulated pretty smoothly, by the way... the squares on the ground should be circular rings!), and the way the rest of the environment is sampled makes it to "look like" the world is tilted. 

I'll add a workaround for cases where motion is known in advance to be planar...

jlblancoc, 2025-02-15:
It's done via a new env var `MOLA_NAVSTATE_ENFORCE_PLANAR_MOTION`: 

```bash
MOLA_NAVSTATE_ENFORCE_PLANAR_MOTION=true \
MOLA_ABS_MIN_SENSOR_RANGE=200 \
MOLA_SIGMA_MIN_MOTION=0.5 \
MOLA_MINIMUM_RANGE_FILTER=1.0 \
MOLA_LIDAR_TOPIC=/lidar/points \
mola-lo-gui-rosbag2 rosbag2_2025_02_13-10_19_34_0.mcap
``` 

You need to git pull the repos: 
- mola_state_estimation
- mola_lidar_odometry

and rebuild. 

I also added the ros2 launch flag `enforce_planar_motion:=True` to active this.

jlblancoc, 2025-02-19:
Hope it's fixed with this new flag... closing; reopen if needed.

closed 2025-02-19 by jlblancoc