MOLAorg/mola_lidar_odometry #19: Odometry Not Updating in Certain Scenario

marmik18, 2025-02-15:
The Odometry pose is not updating even though the ICP quality is above 90%, mainly after the frame number 5000.

You can download the ROS 2 bag from [here](https://we.tl/t-QQACxrvaOQ), the link is valid until 22 Feb 2025.

[image]

I mostly think this is because the environment is very similar, but I can see the changes in wheel odometry on `/odom` topic.
Please note, this is a simulated environment.

[image]

Can you please guide on the configuration that I should change in order to make it work in such an environment?

marmik18, 2025-02-20:
Hey @jlblancoc , 
I have been able to get some things working after tweaking certain parameters.
I don't know whether that's the right way to go. Hope you can share your views on it.

Parameter that I have tweaked:
``` bash
# Environment Variables
MOLA_SIGMA_MIN_MOTION=0.5
MOLA_MINIMUM_RANGE_FILTER=2.0
```

```yaml
# in lidar3d-default.yaml 
range_max: 10.0
```
I don't know why changing `ESTIMATED_SENSOR_MAX_RANGE` env variable did not work for me.

After changing these parameters, I am able to get relatively good lidar odometry and the local map is also updating.

jlblancoc, 2025-02-20:
I didn't have bandwidth to study your data yet, but it make sense that tweaking those params may be beneficial in some cases: 
- MOLA_MINIMUM_RANGE_FILTER: The default is just a heuristic for car-like datasets. For small robots the default formula here may leave many near obstacles out of sight, while they are actually useful. The goal of this parameter is just to remove the robot body, and part of the too-close ground points since that's normally not too useful.
- MOLA_SIGMA_MIN_MOTION: This is the lower limit for the dynamic "sigma" (refer to the MOLA-LO 2025 IJRR paper, or the KISS-ICP paper). Smaller values tend to lead to more stable & accurate results. But when subject to abrupt motions, we need "sigma" to quickly grow to "catch up" with pairings lidar vs local map which may be farther away than when moving smoothly. A trade-off, and subtle to trial and error...

The problem with ESTIMATED_SENSOR_MAX_RANGE sounds like a potential bug... if you find the reason, feel free of reporting and even better opening a PR! 😄

marmik18, 2025-02-21:
It seems that `ESTIMATED_SENSOR_MAX_RANGE` is updated in `updatePipelineDynamicVariables()` on this [line](https://github.com/MOLAorg/mola_lidar_odometry/blob/878717137cd19aedd918bb12530172f3f5b0cb0f/module/src/LidarOdometry.cpp#L1724C8-L1724C34). It seems that the variable is dynamic, and updated on every iteration. If this is a design choice, would it be better to tune the multiplier with an environment variable?

```diff
# lidar3d-default.yaml
  - class_name: mp2p_icp_filters::FilterByRange
    params:
      input_pointcloud_layer: 'decimated_for_map_raw'
      output_layer_between: 'decimated_for_map_by_range'
      range_min: '${MOLA_MINIMUM_RANGE_FILTER|max(1.0, 0.03*ESTIMATED_SENSOR_MAX_RANGE)}'//
-      range_max: 1.2*ESTIMATED_SENSOR_MAX_RANGE
+      range_max: ${MOLA_LOCAL_MAP_FILTER_BY_RANGE_MAX|1.2}*ESTIMATED_SENSOR_MAX_RANGE
```

jlblancoc, 2025-03-03:
Hey, 
Sure, `ESTIMATED_SENSOR_MAX_RANGE` is a dynamic variable by design, sorry, I just did read too quickly last time. 

There's already a `MOLA_ABS_MIN_SENSOR_RANGE` in `lidar3d-default.yaml` which can be used to tune it, and of course, if needed in some particular case, one could modify the line you pointed out. 

`ESTIMATED_SENSOR_MAX_RANGE` is not only used for that FilterByRange filtering of the robot body, but for automatically guessing the map voxel resolution, etc.

closed 2025-03-03 by jlblancoc