MOLAorg/mola_lidar_odometry #20: Colcon Build Error: mola_lidar_odometry stderr

architjain19, 2025-03-11:
Getting this error while building the mola ros2 workspace. Rest all 27 packages built successfully. But got stuck with this error for `mola_lidar_odometry` package.

> Refering this source for installing from source: https://docs.mola-slam.org/latest/

```                                                  
--- stderr: mola_lidar_odometry                               
/home/arc-hitter/ros2_mola_ws/src/mola_lidar_odometry/module/src/LidarOdometry_Publish.cpp: In member function ‘void mola::LidarOdometry::publishMetricMapGeoreferencingData()’:
/home/arc-hitter/ros2_mola_ws/src/mola_lidar_odometry/module/src/LidarOdometry_Publish.cpp:167:22: error: ‘struct mola::MapSourceBase::MapUpdate’ has no member named ‘georeferencing’
  167 |   auto & georef = mu.georeferencing.emplace();
      |                      ^~~~~~~~~~~~~~
/home/arc-hitter/ros2_mola_ws/src/mola_lidar_odometry/module/src/LidarOdometry_Publish.cpp: In member function ‘void mola::LidarOdometry::onPublishDiagnostics()’:
/home/arc-hitter/ros2_mola_ws/src/mola_lidar_odometry/module/src/LidarOdometry_Publish.cpp:196:3: error: ‘DiagnosticsOutput’ was not declared in this scope
  196 |   DiagnosticsOutput diag;
      |   ^~~~~~~~~~~~~~~~~
/home/arc-hitter/ros2_mola_ws/src/mola_lidar_odometry/module/src/LidarOdometry_Publish.cpp:197:3: error: ‘diag’ was not declared in this scope
  197 |   diag.timestamp = curStamp;
      |   ^~~~
/home/arc-hitter/ros2_mola_ws/src/mola_lidar_odometry/module/src/LidarOdometry_Publish.cpp:201:3: error: ‘module_publish_diagnostics’ was not declared in this scope; did you mean ‘onPublishDiagnostics’?
  201 |   module_publish_diagnostics(diag);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~
      |   onPublishDiagnostics
/home/arc-hitter/ros2_mola_ws/src/mola_lidar_odometry/module/src/LidarOdometry_Main.cpp: In member function ‘virtual void mola::LidarOdometry::spinOnce()’:
/home/arc-hitter/ros2_mola_ws/src/mola_lidar_odometry/module/src/LidarOdometry_Main.cpp:130:7: error: ‘module_is_time_to_publish_diagnostics’ was not declared in this scope
  130 |   if (module_is_time_to_publish_diagnostics()) {
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gmake[2]: *** [CMakeFiles/mola_lidar_odometry.dir/build.make:188: CMakeFiles/mola_lidar_odometry.dir/module/src/LidarOdometry_Publish.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[2]: *** [CMakeFiles/mola_lidar_odometry.dir/build.make:132: CMakeFiles/mola_lidar_odometry.dir/module/src/LidarOdometry_Main.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:126: CMakeFiles/mola_lidar_odometry.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< mola_lidar_odometry [5.30s, exited with code 2]

```

Kindly help in figuring out this issue

Regards,
Archit J

r-aguilera, 2025-03-11:
Copied and updated all the packages in a new workspace. The stack compiles for me without such stderr messages. Was this a freshly created workspace or a previous one? My first guess is your `mola` package needs to be updated, as [DiagnosticsOutput](https://github.com/MOLAorg/mola/commit/4d085998a1b1b2786ee081894b56e96c41b91efa) and [Georeferencing](https://github.com/MOLAorg/mola/commit/5c42d0bace7f4943cc492a11b8fc36b307f45819) were relatively recently added structs.

architjain19, 2025-03-12:
I tried creating a new workspace and clone all repositories mentioned in the documentation

```
mkdir -p ~/ros2_mola_ws/src/
cd ~/ros2_mola_ws/src/

# Main MOLA modules:
git clone https://github.com/MOLAorg/mola_common.git
git clone https://github.com/MOLAorg/mp2p_icp.git --recursive
git clone https://github.com/MOLAorg/mola.git --recursive
git clone https://github.com/MOLAorg/mola_state_estimation.git
git clone https://github.com/MOLAorg/mola_test_datasets.git

# MOLA lidar odometry package:
git clone https://github.com/MOLAorg/mola_lidar_odometry.git --recursive
```

Do i need to have any specific version of installation?

jlblancoc, 2025-03-12:
@architjain19 That build error means the "mola_kernel" you are trying to build against is >=1.6.1 but an older than when that structure was introduced...

It might be that you have an older version of any of the packages, conflicting if installed via "apt"?

check with: 

```
dpkg --list | grep mola
```

If any line appears starting with "ii" it means you have them via "apt". Remove them and try with cloned sources only. 

Double check, via "git pull", "git status" that your sources for "MOLAorg/mola" are really in sync with the "develop" version.

jlblancoc, 2025-03-12:
PS: When updating MOLAorg/mola via "git pull", in order for the other packages to realize of the new version we need to re-invoke cmake. The easiest way is to just delete all `ros2_ws/build/*` directories (there are finer ways, but that's the easiest...)

jlblancoc, 2025-04-19:
Seems stalled. Please, reopen if needed.

closed 2025-04-19 by jlblancoc