MOLAorg/mola_lidar_odometry #21: Resources

Corund060, 2025-05-12:
Using mola lidar odometry localization on jetson xavier uses 90% and up of CPU (together with lidar node).
I get ''LiDAR worker thread too busy' warning frequently.
With the standard parameters.
What are the first params to look at when trying to lower this?

jlblancoc, 2025-05-13:
Hi @Corund060 ! 

Well, first, note that normally 90% CPU means it's using "just" 90% of one core in multicore systems, so the overall CPU usage might be >100% and still be responsive. 

That said, you can try experimenting setting these environment variables, which control the [default 3D LiDAR pipeline](https://docs.mola-slam.org/latest/mola_lo_pipelines.html#pipelines-env-vars): 
- `MOLA_LOCAL_VOXELMAP_RESOLUTION`: Try with `0.5` or `1.0` (meters).
- `MOLA_ICP_CLOUD_DECIMATION` and `MOLA_MAP_CLOUD_DECIMATION`: Try setting the "icp" value ~ 3 times the "map" value for consistency. Larger values will be faster, but lower resolution maps.

How to set these env vars: 

1) Export: 

```bash
export MOLA_LOCAL_VOXELMAP_RESOLUTION=1.0
# here, your regular "mola-cli", "ros2 launch", etc. command to use MOLA
``` 

2) Just volatile env vars:

```bash
MOLA_LOCAL_VOXELMAP_RESOLUTION=1.0 \
# here, your regular "mola-cli", "ros2 launch", etc. command to use MOLA
``` 

Hope it helps.

Corund060, 2025-05-14:
Hi @jlblancoc 

Thank you for your answer.

90+% is an average across 6 cores.

I'm passing Mola my own copy of lidar3d-default.yaml so I don't need to export those vars.
According to the formula voxel_size was already at about 0.8 so I'll try increasing it more as I understanding making it 0.5 would be worse.

I am also using Mola on humble form ros-humble-mola-lidar-odometry, maybe it was significantly optimized since then ?

Is decreasing any of these worth a try:
` 
  params:
    maxIterations: 300
    minAbsStep_trans: 1e-4
    minAbsStep_rot: 5e-5
`