MOLAorg/mola_lidar_odometry #22: Unable to change mola_lo_reference_frame

PedroS235, 2025-05-16:
Hello,

when I set `mola_lo_reference_frame` to anything else other than `map` the program crashes, and in specific it seems to be related to the StateEstimation. Is there any reason for this?

To give you why I want to change this value, is that I am trying out mola_lidar_odometry on a handheld device, and therefore I would like to have it map to odom instead, since I don't have an odom source.

```sh
[mola-cli-1] [10:44:03.1516|ERROR|MolaLauncherApp] Error: Will shutdown since thread for module named `state_estimation` ended due to an exception:
[mola-cli-1] ==== MRPT exception ====
[mola-cli-1] Message:  Requested results in unknown frame_id: 'map'
[mola-cli-1] Location: ./src/StateEstimationSmoother.cpp:786: [std::optional<mola::NavState> mola::state_estimation_smoother::StateEstimationSmoother::build_and_optimize_fg(mrpt::Clock::time_point, const string&)
[mola-cli-1] Call stack backtrace:
[mola-cli-1] [0 ] (unknown file) mola::state_estimation_smoother::StateEstimationSmoother::build_and_optimize_fg(std::chrono::time_point<mrpt::Clock, std::chrono::duration<long, std::ratio<1l, 10000000l> > >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[mola-cli-1] [1 ] (unknown file) mola::state_estimation_smoother::StateEstimationSmoother::spinOnce()
[mola-cli-1] [2 ] (unknown file) mola::MolaLauncherApp::executor_thread(mola::MolaLauncherApp::InfoPerRunningThread&)
[mola-cli-1] [3 ]     0xffff7f9c31fc
[mola-cli-1] [4 ]     0xffff7f78d5b8
[mola-cli-1] [5 ]     0xffff7f7f5edc
[mola-cli-1] [10:44:03.3497|INFO |MolaLauncherApp] Main SLAM/localization loop ended.
[mola-cli-1] [10:44:03.3497|INFO |MolaLauncherApp] shutdown(): Shutting down 3 module threads...
```

jlblancoc, 2025-05-17:
Ok, I could reproduce this with: 

```bash
MOLA_STATE_ESTIMATOR_YAML=../state-estimator-params/state-estimation-smoother.yaml  \
MOLA_STATE_ESTIMATOR=mola::state_estimation_simple::StateEstimationSmoother \
MOLA_LO_PUBLISH_REF_FRAME=/map2  \
ros2 launch mola_lidar_odometry kitti-ros2-lidar-odometry.launch.py
```

I'll dig into it. 

PS: @PedroS235 does the same happens if you use the "simple" state estimator instead?

PedroS235, 2025-05-17:
Hi @jlblancoc ,

Yes, with both estimators, I got the same issue.

I am not entirely sure, but on the previous release, this did not happen.

If I can provide anything that can help you, let me know.

jlblancoc, 2025-05-19:
Progress: After the latest commit in this repo, there is now an env var `MOLA_TF_MAP`  that allows changing the reference frame in the smoother state estimator too: 

```bash
MOLA_TF_MAP=/map2 \
MOLA_STATE_ESTIMATOR_YAML=../state-estimator-params/state-estimation-smoother.yaml  \
MOLA_STATE_ESTIMATOR=mola::state_estimation_simple::StateEstimationSmoother \
MOLA_LO_PUBLISH_REF_FRAME=/map2  \
ros2 launch mola_lidar_odometry kitti-ros2-lidar-odometry.launch.py
```

Right now, it doesn't crash, and it moves, but the estimator does something wrong that needs investigation: 

```
[mola-cli-1] [10:38:36.4049|WARN |StateEstimationSmoother] A constant-velocity kinematics factor has been added for a dT=41.992 s.
```

@PedroS235  Can you please provide an example call that makes it to fail using the simple state estimator (the default one)?

PedroS235, 2025-05-20:
That's perfect! Thanks so much for the fast responses. I will try to run this by the end of the week. You can close the issue in the meantime if you desire so.